5G-ERA/middleware #48: Dashboard endpoints

Artonus, 2022-11-30:
The dashboard endpoints will cover extracting specific data from the Middleware APIs.

1. The endpoint has to present information about the cloud and the edges they're currently operating in the system.
The endpoint has to return the following properties:

    Last Time the location updated
    Name of the Edge/Cloud
    Is it online
    Is it busy
    Number of containers it runs
    status of a cloud/edge

2. The endpoint has to present information about the robots and the tasks that they're currently executing.
The endpoint has to return the following properties:

 *   Robot name/id
 *   Task name/id
 *   Date and Time it started
 *   Status of a task
 *   The date and Time finished
 *   status of a robot

adrianLIrobotics, 2022-12-01:
Edited issue and added link to Dashboard proposal for feature tracking.

adrianLIrobotics, 2022-12-15:
I will continue the extra endpoints as agreed.

adrianLIrobotics, 2022-12-21:
Added Basic cascading grid for action sequence endpoint for dashboard.

adrianLIrobotics, 2022-12-27:
Added missing endpoints, only needs QA testing

adrianLIrobotics, 2023-01-09:
issue #93 and #94 still under progress

closed 2023-01-18 by Artonus