5G-ERA/middleware #54: Integration of new netApp design with middleware and AS/AC

adrianLIrobotics, 2022-12-01:
**Describe the solution you'd like**
The new netApp design requires the endpoint for planning in middleware to pass the ROS topics and topics types. This is neccesary for querying for suitable netApp's based on the requried input topics. This interface to the netApp requires to be loaded in redis DB index of netApp as extra information that must be included.
Additionally, the AC/AS needs to be expaned to include in the ros goal the topics of the robot. AS will be integrated with the netApp client. 

**Describe alternatives you've considered**
N/A

Artonus, 2023-04-14:
What needs to be discussed is the automatic gateway reconfiguration #134. It will allow for the redirection of the GStreamer traffic directly to the NetApp through a unified access point.

Artonus, 2023-06-14:
Blocked by #134 and #158

Artonus, 2023-07-25:
AC/AS needs to embed the Network Application client for it to handle the connection to the Middleware. On top of this, the AC/AS will serve it's own designed purposes of keeping the interaction between robot and the Middleware

Artonus, 2023-10-11:
This issue is out of date. Now we natively support ROS1 and ROS2 applications.

closed 2023-10-11 by Artonus