5G-ERA/middleware #58: Task GUIDs needs to be descriptive and not a random number

adrianLIrobotics, 2022-12-01:
**Describe the bug**
For ROS developer it is very difficult to remember what the task GUID is about by looking at the GUID. It should be a task name and will be descriptive.

It must be integrated with the middleware correctly and with the Redis database.

**Additional context**
Reported on RobotNik september intgration 2022.

Artonus, 2022-12-02:
As talked through with Michal, the Robot will call the Action Client with two possible requests:

1. Direct call for a single NetApp, robot will specify what task it wants to conduct
2. Semantic planning call, the robot will call the Middleware with the request to prepare the high-level task and the resources needed to conduct the task.

adrianLIrobotics, 2022-12-06:
Isnt this issue duplicate #79 ?

adrianLIrobotics, 2022-12-15:
closing as duplicate

closed 2022-12-15 by adrianLIrobotics