5G-ERA/middleware #67: Resource planner 5g slicing modeling and inclusion in planning

adrianLIrobotics, 2022-12-01:
**Is your feature request related to a problem? Please describe.**
5G slicing support for middleware in terms of modeling in redis, in middleware and how planner and dedicated 5g policies apply.

**Additional context**
Add any other context or screenshots about the feature request here.
This is required for the next integration meeting in 2023. Get input from networking and 5g partners to model this.

adrianLIrobotics, 2022-12-15:
[5gSlicing.docx](https://github.com/5G-ERA/middleware/files/10237947/5gSlicing.docx)

Artonus, 2023-05-05:
Moved to v0.5 Milestone, as it will be a crucial part of testing the testbed and slice integration.

The slices provided are:
* default
* ultra-reliable and low latency (URLLC)
The assignment of the slice could be controlled by policies. For example, we could have Instance-scoped policies that would alter the planning process for Instances. If the URLLC network policy is specified for the Instance, the Resource Planner will ask Slice Manager to assign the specific robot to the URLLC Slice.

This issue is associated with the `Policies` development (#124)

closed 2023-06-01 by Artonus