5G-ERA/middleware #68: Switch over extension work

adrianLIrobotics, 2022-12-01:
**Describe the solution you'd like**
The middleware should be able to allow the robot to use the cloud/edge switch over if a proactive replan is triggered. Switchover is also required for 5G slicing management and usage.  Check what are the implications from the planning perspective.

**Additional context**
Add any other context or screenshots about the feature request here.

Artonus, 2022-12-02:
[image]
Attached is the image presenting the conceptual idea behind the switchover

Artonus, 2022-12-05:
# hahahaha

Artonus, 2022-12-06:
The Edge switchover should be handled in multiple steps:

1. Provide a MessageBus to send the requests to other instances of the Middleware
2. Working on a single location, change the connectivity between the TaskPlanner and Orchestrator to use MessageBus
3. Send the deployment requests to the single (local) Edge
4. Plan for work in a different location and send the request to the new Edge to deploy the desired resources
5. Define a way in which the Robot will request the switchover (possible connection with #51)
6. Specify a path in the Middleware to request the change to the specified location
7. Send the requests to deploy the desired resources from one Edge to another Edge through MessageBus
8. Upon receiving the confirmation that the request has been successfully processed and the service has been redeployed to a new location, delete the service from the current site (the action plan of the Robot should be updated automatically with the new address)
9. Inform the Robot that the switchover process has been completed and that it can use the new services

closed 2023-03-16 by Artonus