5G-ERA/middleware #69: Check how to run lua querys with parameters from within the baseModel in middlware

adrianLIrobotics, 2022-12-01:
**Is your feature request related to a problem? Please describe.**
For complex query, we may need to adjust the baseModel to accomodate parameters to the lua query system. This is low priority.

**Describe the solution you'd like**
N/A

**Describe alternatives you've considered**
N/A

**Additional context**
No real usecase identified yet.

Artonus, 2022-12-02:
As an alternative to using Lua for complex queries, we could use another database system. Of course, from the scalability point of view, some document databases would be more suitable for us. But the relational database that provides good replication would also work.

adrianLIrobotics, 2022-12-15:
We will be using: https://github.com/redis/redis-om-dotnet

adrianLIrobotics, 2022-12-15:
I am closing therefore this issue if ok

closed 2022-12-15 by adrianLIrobotics