5G-ERA/middleware #80: Define the testing template

Artonus, 2022-12-06:
The testing templates should be defined and distributed across the team. In addition, the testing scenarios should be prepared for the testing of the Issues and QA. 

The testing templates will be used not only by the QA but also by the team itself to initially test the produced features.

adrianLIrobotics, 2022-12-12:
- [ ] Check University VPN for accesing testing scenario.

diwalipathak, 2022-12-22:
Test template ; 
Tester name 
Test date
Feature tested 
Project/test cycle
test platform
device/browser 
test purpose
[Test-Template.xlsx](https://github.com/5G-ERA/middleware/files/10286152/Test-Template.xlsx)

adrianLIrobotics, 2023-01-11:
How are we doing with the testing enviroment? VPN needs to be provided.

closed 2023-02-21 by Artonus