5G-ERA/middleware #86: Robot task basic ontology

adrianLIrobotics, 2022-12-15:
**Research Description**
A clear and concise description of the subject to be researched.
Robot task names need to be descriptive and not Guids.

**Is the research related to an already known bug or feature?**
Provide the tag for the bug or feture. #79 

**Research related branch**
Robotics ontology

**Is the research directly related to an existing system component?**
RedisInterface API, Orchestrator API, ML Tool, AWS, etc

**Research summary**
Describe your findings

**Pros**
List the potential benefits 

**Cons**
List the potential drawbacks

**References**
List the used sources

diwalipathak, 2022-12-15:
Industry           |
|--------------------|
| Chemical           |Task Go for a safety inspection in Lab_1
| Commerce           |Task perform a customer service-chatbot questionnaire
| Construction       |Task Move certain object (brick)
| Education & Research         |Task Tell a student about Robotics 
| Public service     |Task Prepare necessary document
| Transport          |Task  Carry a particular good (vacuum cleaner) inside
| Textile            |Task Draw a print design on fabric
| Postal and telecom |Task Putting stamp on the post
| Energy             |Task Repair the pipeline 
| Media & Culture    |Task Read the mood (user by expressions) face recognition 
| Mining             |Task Remove the stone 
| Tourism and hotel   |Task Find the best route to get the Hotel#
| Healthcare         |Task Give medicine to the patient 
| forestry           |Task Find the best route to get to the injured animal (lion)
| food/drink         |Task Count the pizza boxes to be delivered 
| metal production   |Task Assemble the metal parts of the product

diwalipathak, 2022-12-15:
[Industry.docx](https://github.com/5G-ERA/middleware/files/10236554/Industry.docx)

adrianLIrobotics, 2022-12-15:
Added proposal: https://github.com/5G-ERA/middleware/blob/main/docs/Proposals/RobotTaskOntology.md

adrianLIrobotics, 2023-01-03:
Add work done with Renxi over basic ontology

adrianLIrobotics, 2023-01-06:
[RobotTaskOntology.md](https://github.com/5G-ERA/middleware/files/10359785/RobotTaskOntology.md)

adrianLIrobotics, 2023-01-06:
[RobotTaskOntology2.md](https://github.com/5G-ERA/middleware/files/10359796/RobotTaskOntology2.md)

adrianLIrobotics, 2023-01-06:
[robot_ontology.txt](https://github.com/5G-ERA/middleware/files/10359800/robot_ontology.txt)