5G-ERA/middleware #100: Update public roadmap of project

adrianLIrobotics, 2023-01-15:
**Is your feature request related to a problem? Please describe.**
Currently we have in the main git page a public roadmap. It should be updated every month (with agreed features of team members) and maybe also updated in the main web page. Contact EBos for this one. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

Artonus, 2023-02-09:
The public roadmap of the documentation should be adjusted to the current state of the release planning to give a very general overview of what is currently conducted in the development of the 5G-ERA Middleware

diwalipathak, 2023-03-20:
done !

closed 2023-03-27 by Artonus