5G-ERA/middleware #97: Prepare onboarding templates of edges, cloud, instance for dashboard

adrianLIrobotics, 2023-01-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

The dashboard interface has a button that can be used to download the onboarding templates for entities in middleware. This template is completed by dashboard user and can be uploaded to the dashboard to create a new entity in middleware.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Artonus, 2023-02-02:
We have to simplify the models for passing and retrieving the data. 
At the moment the models have unnecessary p[properties like Relations and other values that should not be specified by the end user.

This will be a breaking change when it comes to the Dashboard, so the partners will have to be informed accordingly.

For release v0.2, we have to prepare the model structure about which we will inform partners. The Implementation of the new models will be scheduled for release v0.3.

Artonus, 2023-02-20:
@adrianLIrobotics prepared the onboarding templates. Now, for the v0.3, it is time to prepare and document changes and let partners know their handling process

Artonus, 2023-02-27:
On top of this, I suggest adding model validation when requests are made. 
The package that will help this is https://docs.fluentvalidation.net/en/latest/index.html

closed 2023-04-13 by Artonus