5G-ERA/middleware #124: Working policies implementation required

adrianLIrobotics, 2023-03-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Implement, design and validate all middleware policies.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Artonus, 2023-03-31:
We will have to consider not only global policies but also scoped policies. During the meeting, we identified the example "Deployment Policy" that will allow the deployment of the specific NetApp only in the private Cloud/Edge machine, not the public one.

Artonus, 2023-05-05:
The initial policy in the system will be the Network Slice Selection Policy. This is needed as part of the Slice Manager integration into the Middleware. 
The policy can be attached to the `Instance` and specify what actions must be considered when deploying the `Instance`.  
The example `Policy` _URLLC Network_ would mean that if the Slicing mechanism is available, connect the `Robot` utilising the instance to the URLLC Network Slice.

Things to be compleated:
* [ ] Division of the `Policies` into the system-scoped and Instance-scoped
* [ ] How to attach policies to the Instance / another object like `Robot`.
* [ ] Implementation of the URLLC `Policy`.

PR #155 (2023-06-01): 5G Location selection Policies implementation
# Description

Operational policy implementation is aimed to allow for the customization of the deployment and location planning process but not only. The Policies aim to allow to change the system's behaviour in a user-controlled manner. 
The 2 Policy types were distinguished:
* `system`-scoped - applicable to the whole system functioning
* `resource`-scoped - applicable to the specific resource like a robot or the Network Application instance

The Policies available in the Middleware will be predefined. Their operational properties will be user m [... cut]
closed 2023-06-01 by Artonus