5G-ERA/middleware #145: Infrastructure as code - automated infrastructure deployment

radu-popescu, 2023-04-28:
**Is your feature request related to a problem? Please describe.**
The need for an automated infrastructure deployment has to be addressed for to purpose of quick recovery in case of potential incidents on the AWS platform.

**Describe the solution you'd like**
Terraform infrastructure as code

**Describe alternatives you've considered**
AWS Cloud Formation, Ansible 

**Additional context**
This has to be implemented as soon as possible, it is a high priority task.

Artonus, 2023-04-28:
@radu-popescu we already have an issue created for this -  #116. Please add the details over there if necessary, I will close this issue as duplicate.

closed 2023-04-28 by Artonus

radu-popescu, 2023-04-28:
My bad, I haven't seen that we already had one open.

Artonus, 2023-04-28:
No problem :)